nodemcu/nodemcu-devkit-v1.0 #1: File writes fail all the time

rjpcomputing, 2015-06-16:
Hi,

I can not save files to the file system on my board. I have tried the last 4 firmware releases and it doesn't change the problem. It seems to be related to the  amica boards. I have the revision before this and it works 100% of the time.

```lua
file.open("script1.lua","w+")
file.writeline([[print( "Hello World" )]])
file.close()
```
**OUTPUT**:
```
stdin:1: open a file first
```

I am opening the file on the first line.

Please help

vowstar, 2015-06-17:
Hello, you may use old flashtool.
Options:
1. Please use latest esptool.py and with DIO option to flash firmware. 
https://github.com/themadinventor/esptool
2. Please use latest NodeMCU flasher with default option (must click 'restore to default' in advanced tab)
https://github.com/nodemcu/nodemcu-flasher
3. Please use latest Espressif's flash tool(without auto download support)
If always have problem, please use latest flash download tool from espressif. http://bbs.espressif.com/viewtopic.php?f=5&t=433 Please use DIO mode and 32M flash size option, and flash latest firmware to 0x00000. Before flashing firmware, please hold FLASH button, and press RST button once. When our firmware download tool released, it will flash firmware automatically and needn't press any button.

rjpcomputing, 2015-06-17:
Hi,

I have been using `esptool.py` downloaded from the master branch (e96336f6561109e67afe03c0695d1e5b0de15da6) and it did not work.

Here is the commandline I am using:
```
./esptool.py --port /dev/ttyUSB0 write_flash 0x00000 <firmware> -fs 32m -fm dio -ff 40m
```
I am using the Amica black board on Linux.

Air000, 2015-06-26:
I have the same problem with @rjpcomputing , also using the Amica black board. 
I tried all the methods. Only the tool on http://bbs.espressif.com/viewtopic.php?f=5&t=433 work fine.
But I need esptool.py on Linux

leonciokof, 2015-07-13:
@Air000 how to use esptool from http://bbs.espressif.com/viewtopic.php?f=5&t=433 ?

Air000, 2015-07-17:
@leonciokof 
my esptool.py is download from https://github.com/themadinventor/esptool/blob/master/esptool.py.
and I test the esptool from  http://bbs.espressif.com/viewtopic.php?f=5&t=433, also output nothing.

closed 2015-07-23 by vowstar

rjpcomputing, 2015-07-23:
@vowstar Why is this closed? The bug is not fixed. Should this be reported on a different tracker?

I don't think that Wine + flash_download_tool.exe is a good enough solution when a feature of NodeMCU is cross-platform.

Thoughts?

rjpcomputing, 2015-07-23:
I think esptool might have got a fix. I have yet to test it, but the commit looks like it should work.
https://github.com/themadinventor/esptool/commit/043f9287c6f7d8a8c2579395f790629f3d85683c

Has anyone tried esptool after July 8th?

rjpcomputing, 2015-08-03:
Just to let other know I was able to use the latest esptool (>July 8th) to flash the board.