nodemcu/nodemcu-devkit-v1.0 #2: CP2102 OSX drivers?

MacDada, 2015-07-08:
Where do I find **working** drivers for OSX 10.10 (Yosemite)?

1. I downloaded OSX drivers from [the vendor website](https://www.silabs.com/products/mcu/Pages/USBtoUARTBridgeVCPDrivers.aspx): [direct zip download](https://www.silabs.com/Support%20Documents/Software/Mac_OSX_VCP_Driver.zip)
2. Unpacked the ZIP, mounted DMG file, installed using the PKG installer.
3. Restarted computer
4. I connected NodeMcu devkit v1.0 through USB 3.0 port

No serial port available for the device.

```
$ ls -lah /dev/cu*
crw-rw-rw-  1 root  wheel   18,   1 Jul  8 13:32 /dev/cu.Bluetooth-Incoming-Port
crw-rw-rw-  1 root  wheel   18,   3 Jul  8 13:32 /dev/cu.Bluetooth-Modem
crw-rw-rw-  1 root  wheel   18,   5 Jul  8 13:32 /dev/cu.DadaFon4-WirelessiAP

$ ls -lah /dev/tty.*
crw-rw-rw-  1 root  wheel   18,   0 Jul  8 13:32 /dev/tty.Bluetooth-Incoming-Port
crw-rw-rw-  1 root  wheel   18,   2 Jul  8 13:32 /dev/tty.Bluetooth-Modem
crw-rw-rw-  1 root  wheel   18,   4 Jul  8 13:32 /dev/tty.DadaFon4-WirelessiAP
```

No success with ESPlorer and Arduino IDE.

MacDada, 2015-07-08:
@squix78 Apparently you've got this working – any details?

Did it just work OOTB? What OSX version? What model of Mac?

Do you see the kext being loaded in `kextstat`? (not sure if this should show it…)

marcoskirsch, 2015-07-08:
SiLabs' drivers worked out of the box for me (OS X Yosemite). They are signed so there's no need to turn off checks in the OS.

There's some issues I've had and I cannot say for sure if they are due to the drivers: I can use NodeMCU devkit 1.0 with screen and ESPlorer in general but I was not able to make it work with mini-com nor have I been able to flash it with esptool. These things worked for me onESP01 and different USB-serial adapter.

MacDada, 2015-07-08:
@marcoskirsch Could you please paste what `kextstat` shows you? Is the driver listed there? What's your exact OSX/Mac?

marcoskirsch, 2015-07-08:
OS X Yosemite 10.10.4 (14E46)
MacBook Pro (Retina, 13-inch, Early 2015) 3.1 GHz Intel Core i7

kextstat (scroll to the bottom):
```
Index Refs Address            Size       Wired      Name (Version) <Linked Against>
    1   89 0xffffff7f80a0f000 0x8c50     0x8c50     com.apple.kpi.bsd (14.4.0)
    2    7 0xffffff7f80e66000 0x28c0     0x28c0     com.apple.kpi.dsep (14.4.0)
    3  112 0xffffff7f80a1d000 0x20500    0x20500    com.apple.kpi.iokit (14.4.0)
    4  120 0xffffff7f80a03000 0xbf50     0xbf50     com.apple.kpi.libkern (14.4.0)
    5  103 0xffffff7f80a00000 0x2d50     0x2d50     com.apple.kpi.mach (14.4.0)
    6   49 0xffffff7f80a3e000 0x8590     0x8590     com.apple.kpi.private (14.4.0)
    7   68 0xffffff7f80a18000 0x4900     0x4900     com.apple.kpi.unsupported (14.4.0)
    8    4 0xffffff7f80e6c000 0xa0000    0xa0000    com.apple.kec.corecrypto (1.0) <7 6 5 4 3 1>
    9    0 0xffffff7f80fa4000 0xc000     0xc000     com.apple.kec.pthread (1) <7 6 5 4 1>
   10    1 0xffffff7f80fd4000 0x9000     0x9000     com.apple.kec.Libm (1) <4>
   11   24 0xffffff7f8140f000 0x9000     0x9000     com.apple.iokit.IOACPIFamily (1.4) <7 6 4 3>
   12   30 0xffffff7f80a47000 0x2f000    0x2f000    com.apple.iokit.IOPCIFamily (2.9) <7 6 5 4 3>
   13    2 0xffffff7f829e5000 0x5f000    0x5f000    com.apple.driver.AppleACPIPlatform (3.1) <12 11 7 6 5 4 3 1>
   14    1 0xffffff7f81125000 0xb000     0xb000     com.apple.driver.AppleFDEKeyStore (28.30) <8 7 6 5 4 3 1>
   15    2 0xffffff7f81408000 0x7000     0x7000     com.apple.iokit.IOReportFamily (31) <5 4 3>
   16    7 0xffffff7f80b2d000 0x26000    0x26000    com.apple.iokit.IOStorageFamily (2.0) <7 6 5 4 3 1>
   17    0 0xffffff7f8150f000 0x19000    0x19000    com.apple.driver.DiskImages (397) <16 7 6 5 4 3 1>
   18    1 0xffffff7f8232c000 0xf000     0xf000     com.apple.driver.AppleCredentialManager (1.0) <7 5 4 3 1>
   19    2 0xffffff7f80f71000 0xd000     0xd000     com.apple.driver.AppleMobileFileIntegrity (1.0.5) <7 6 5 4 3 2 1>
   20    0 0xffffff7f8233b000 0x23000    0x23000    com.apple.driver.AppleKeyStore (2) <19 18 8 7 6 5 4 3 1>
   21    0 0xffffff7f80e69000 0x2000     0x2000     com.apple.security.TMSafetyNet (8) <7 6 5 4 2 1>
   22    2 0xffffff7f80f7e000 0x5000     0x5000     com.apple.kext.AppleMatch (1.0.0d1) <4 1>
   23    1 0xffffff7f80f83000 0x17000    0x17000    com.apple.security.sandbox (300.0) <22 19 7 6 5 4 3 2 1>
   24    0 0xffffff7f80f9a000 0x9000     0x9000     com.apple.security.quarantine (3) <23 22 7 6 5 4 2 1>
   25    0 0xffffff7f82a5b000 0x8000     0x8000     com.apple.nke.applicationfirewall (161) <7 6 5 4 3 1>
   26    0 0xffffff7f829b8000 0x3000     0x3000     com.apple.driver.AppleAPIC (1.7) <4 3>
   27    3 0xffffff7f81395000 0x4000     0x4000     com.apple.iokit.IOSMBusFamily (1.1) <5 4 3>
   28    0 0xffffff7f82a4a000 0x7000     0x7000     com.apple.driver.AppleACPIEC (3.1) <27 13 11 5 4 3>
   29    0 0xffffff7f8223f000 0x4000     0x4000     com.apple.driver.AppleSMBIOS (2.1) <7 4 3>
   30    0 0xffffff7f825e8000 0x3000     0x3000     com.apple.driver.AppleHPET (1.8) <11 7 5 4 3>
   31    0 0xffffff7f8226d000 0x8000     0x8000     com.apple.driver.AppleRTC (2.0) <11 5 4 3 1>
   32    5 0xffffff7f81130000 0x76000    0x76000    com.apple.iokit.IOHIDFamily (2.0.0) <14 7 6 5 4 3 2 1>
   33    0 0xffffff7f82a54000 0x4000     0x4000     com.apple.driver.AppleACPIButtons (3.1) <32 13 11 7 6 5 4 3 1>
   34    0 0xffffff7f82245000 0x8000     0x8000     com.apple.driver.AppleSmartBatteryManager (161.0.0) <27 11 5 4 3 1>
   35    1 0xffffff7f82649000 0x5000     0x5000     com.apple.driver.AppleEFIRuntime (2.0) <7 6 5 4 3>
   36    1 0xffffff7f8264e000 0xb000     0xb000     com.apple.driver.AppleEFINVRAM (2.0) <35 7 6 5 4 3 1>
   37    1 0xffffff7f82388000 0xa000     0xa000     com.apple.driver.AppleIntelLpssGspi (2.0.57) <12 11 7 5 4 3>
   39    1 0xffffff7f82369000 0x8000     0x8000     com.apple.driver.AppleIntelLpssDmac (2.0.57) <12 11 5 4 3>
   40   14 0xffffff7f80a7b000 0x65000    0x65000    com.apple.iokit.IOUSBFamily (720.4.4) <12 7 5 4 3 1>
   41    0 0xffffff7f811ce000 0x22000    0x22000    com.apple.driver.AppleUSBXHCI (710.4.11) <40 12 7 5 4 3 1>
   42    1 0xffffff7f82392000 0x14000    0x14000    com.apple.driver.AppleIntelLpssSpiController (2.0.57) <39 37 11 5 4 3>
   43    1 0xffffff7f825bb000 0x17000    0x17000    com.apple.driver.AppleHSSPISupport (41.3) <42 11 5 4 3 1>
   44    6 0xffffff7f810d2000 0x2c000    0x2c000    com.apple.iokit.IONetworkingFamily (3.2) <7 6 5 4 3 1>
   45    1 0xffffff7f8173e000 0x7000     0x7000     com.apple.driver.mDNSOffloadUserClient (1.0.1b8) <44 4 3 1>
   46    1 0xffffff7f81745000 0xab000    0xab000    com.apple.iokit.IO80211Family (730.60) <44 8 7 6 5 4 3 1>
   47    0 0xffffff7f817f8000 0x5e6000   0x5e6000   com.apple.driver.AirPort.Brcm4360 (930.37.3) <46 45 44 12 7 6 5 4 3 1>
   48    5 0xffffff7f81286000 0xde000    0xde000    com.apple.iokit.IOThunderboltFamily (4.2.2) <5 4 3 1>
   49    0 0xffffff7f82154000 0x1b000    0x1b000    com.apple.driver.AppleThunderboltNHI (3.1.7) <48 12 11 5 4 3 1>
   50    2 0xffffff7f820db000 0x24000    0x24000    com.apple.driver.AppleMultitouchDriver (264.6) <32 5 4 3 1>
   51    0 0xffffff7f825d5000 0x9000     0x9000     com.apple.driver.AppleHSSPIHIDDriver (41.3) <50 43 32 5 4 3 1>
   52    0 0xffffff7f82116000 0x6000     0x6000     com.apple.driver.AppleTopCaseHIDEventDriver (67) <50 32 6 5 4 3 1>
[... cut ...]
  143    0 0xffffff7f82ac8000 0xe000     0xe000     com.apple.nke.asp-tcp (8.0.0) <7 6 5 4 3 1>
  144    1 0xffffff7f82ad6000 0x11000    0x11000    com.apple.security.SecureRemotePassword (1.0) <5 4 1>
  145    0 0xffffff7f82ae7000 0x53000    0x53000    com.apple.filesystems.afpfs (11.0) <144 7 6 5 4 3 1>
  191    0 0xffffff7f82b3a000 0x4000     0x4000     com.apple.driver.AppleUSBCDC (4.3.3b1) <40 4 3>
  192    0 0xffffff7f82b3e000 0x6000     0x6000     com.silabs.driver.CP210xVCPDriver (4.1.1) <101 40 4 3>
```

MacDada, 2015-07-08:
@marcoskirsch Thx, you do have the kext loaded… `com.silabs.driver.CP210xVCPDriver (4.1.1)`

The question is: why is it not my case? How do I debug this?

MacDada, 2015-07-08:
http://community.silabs.com/t5/Interface/OS-X-10-10-CP210x-VCP-Driver-Release-Candidate/td-p/138472

```
$ codesign -vvvvd /Library/Extensions/SiLabsUSBDriver.kext/
Executable=/Library/Extensions/SiLabsUSBDriver.kext/Contents/MacOS/SiLabsUSBDriver
Identifier=com.silabs.driver.CP210xVCPDriver
Format=bundle with Mach-O thin (x86_64)
CodeDirectory v=20200 size=497 flags=0x0(none) hashes=17+3 location=embedded
Hash type=sha1 size=20
CDHash=8cc52c4b61556244b230f7e6eed6b18dce4b763d
Signature size=8584
Authority=Developer ID Application: Silicon Laboratories Inc (52444FG85C)
Authority=Developer ID Certification Authority
Authority=Apple Root CA
Timestamp=23 Jun 2015 21:34:41
Info.plist entries=18
TeamIdentifier=52444FG85C
Sealed Resources version=2 rules=12 files=2
Internal requirements count=1 size=228
```

```
$ kextutil -tn /Library/Extensions/SiLabsUSBDriver.kext
/Library/Extensions/SiLabsUSBDriver.kext appears to be loadable (including linkage for on-disk libraries).
```

But it doesn't tell me anything…

abl, 2015-07-08:
Does /dev/cu.SLAB_USBtoUART exist? I'm not sure it creates a tty.

Try the following:

    # Unplug your NodeMCU
    ioreg -w 0 > ~/no-device.txt
    # Plug your NodeMCU in
    ioreg -w 0 > ~/with-device.txt
    diff ~/with-device.txt ~/no-device.txt > ~/diff-device.txt

and upload diff-device.txt as a gist?

There's also a ~known issue with some serial bridges where they have trouble on USB3 - the workaround is to plug in a USB1/USB2 hub in to the USB3 port and then plug your device in to the hub.

MacDada, 2015-07-09:
> Does /dev/cu.SLAB_USBtoUART exist?
@abl It did not.

-------

Anyway, it works. I just had a bad cable (3 of them o.O). After I plugged in a borrowed Samsung phone cable it started working o.O

I guess it's true that same USB cables are only for the power and not every USB cable can exchange data…

----

```
$ ls /dev/tty.*
/dev/tty.Bluetooth-Incoming-Port	/dev/tty.DadaFon4-WirelessiAP
/dev/tty.Bluetooth-Modem		/dev/tty.SLAB_USBtoUART
```

:D

closed 2015-07-09 by MacDada

piejanssens, 2016-02-06:
Exact same problem here. I tried 2 cables already but no SiLabs com port yet.. 
Any way to quickly validate if the cable is the issue or not?

bosschaert, 2016-02-11:
I'm having the same problem on OSX 10.11.3. The cable is definitely good because it works perfectly fine in a Windows VM on the same machine.

adithyaselv, 2016-02-29:
Not working in el capitan 10.11.3 !! :/ works perfectly on windows machine though

kgprocks, 2016-03-25:
I am also facing the exact same issue. It gets detected for the first time and then "error: cannot access /dev/cu.SLAB_USBtoUART" appears. After restart then it again detects and same issue.

hongkongkiwi, 2016-06-02:
Hi Guys,

It looks like this is because the official driver is not signed and in 10.10 & 10.11 they have kext signing :-(

The solution which worked for me is to install this driver from another vendor which is signed, but works just fine! 

http://blog.sengotta.net/signed-mac-os-driver-for-winchiphead-ch340-serial-bridge/

fadushin, 2016-06-09:
I am experiencing the same issue on OS X 10.11.5 (MacBook Pro (Retina, 15-inch, Mid 2014)), using a Wemos D1 Mini and an Amazon USB Cable.

The kernel extension is loaded:

```
> kextstat | grep -i silabs
  151    0 0xffffff7f832ab000 0x6000     0x6000     com.silabs.driver.CP210xVCPDriver (4.10.11) CF83A369-EA70-3E24-B8FE-7351DCF03430 <123 39 4 3>
```

But there is no /dev/tty.SLAB_USBtoUART:

```
> ls tty.*
tty.Bluetooth-Incoming-Port
```
Under System Information the USB device (when attached) shows up as `USB2.0-Serial ` under the `USB 3.0 Bus ` tree:

```
USB 3.0 Bus:

  Host Controller Driver:	AppleUSBXHCILPTH
  PCI Device ID:	0x8c31 
  PCI Revision ID:	0x0005 
  PCI Vendor ID:	0x8086 

...

USB2.0-Serial:

  Product ID:	0x7523
  Vendor ID:	0x1a86
  Version:	2.54
  Speed:	Up to 12 Mb/sec
  Location ID:	0x14200000 / 12
  Current Available (mA):	1000
  Extra Operating Current (mA):	0
```

I have diff'd the results of `ioreg` as suggested above at gist'd at 

https://gist.github.com/fadushin/d4c17e73be26cece8e8cddb6edd704bb

I am disinclined to install an unsigned kernel extension or otherwise not readily downloadable from a reputable vendor, particularly if it is reported to cause kernel panics.

And, hi, @bosschaert !

bosschaert, 2016-06-09:
I installed the Arduino IDE with the ESP8266 addon (as described here https://learn.sparkfun.com/tutorials/esp8266-thing-hookup-guide/installing-the-esp8266-arduino-addon ) and the /dev/cu.SLAB_USBtoUART magically appeared for me. I'm unsure whether it came with the Arduino IDE or with the addon, but its now just working for me.

PS hi @fadushin !

fadushin, 2016-06-10:
Silly me.  My Wemos D1 Mini uses a CH340G chip, not the CP2102 (my son's magnifying glass comes in handy).  Of course that driver won't work.  Using the driver with the right chip actually works.

```
> kextstat | grep -i silabs
  158    0 0xffffff7f83281000 0x6000     0x6000     com.silabs.driver.CP210xVCPDriver (4.10.11) CF83A369-EA70-3E24-B8FE-7351DCF03430 <123 39 4 3>

> ls tty.*
tty.Bluetooth-Incoming-Port	tty.SLAB_USBtoUART
```

NB.  The CH340G driver can be found at

http://www.wemos.cc/downloads/

And instructions are at

http://kig.re/2014/12/31/how-to-use-arduino-nano-mini-pro-with-CH340G-on-mac-osx-yosemite.html

chevdor, 2016-12-08:
If someone runs into such an issue, really **DO** check many cables.
I run into the exact same steps than @MacDada and nothing worked.

I have 2 Mac running Sierra and my ESP8266 works in one and not the other. I could validate a cable to be working on my iMac but it would not on my MBP2016.

I tested randomly many cables, some work, some don´t. Most of those cables usually work fine.

Conclusion: do test many cables....

glauciom, 2017-03-07:
Try to use a USB hub with a power source, like this one: https://i.stack.imgur.com/pzG2u.gif
I had issues when I was trying to use a Presonus Audiobox USB - I've bought this hub, plugged it at the power source and plugged the hardware at the hub and then the USB hub at the Mac book Air's USB port.

XicoBHZ, 2017-04-12:
Guys, can you help me. I've just got my NODEMCU ESP-12e with CP2102 (1.0), I'm using a MAC AIR, but the device doesn't appear on the PORT list, as some of the guys above. I'be tried different cables as well a USB interface that allows me to input an extra 5V DC PSU to feed the NODEMCU, no response at all. The Blink sketch compiles OK, the board is correctly set for NODEMCU 1.0 (ESP-12e Module), but no way to make it works. I heard that I should download a driver for CP2102, is it right? If so, where can I find it? Any help? Thanks!

AddoSolutions, 2017-07-21:
I just experienced this exact same issue.  It ended up taking me a couple hours to figure it out, but I established a highly technical definition of the problem:

**My cable was bad**

🤦‍♂️

jimprince, 2017-09-01:
**_TL;DR: Use a short fat cable_**

I'm going to go ahead and say it is a cable problem, but not a bad cable. I had to try every single one of the dozen or so that I have and I'm not throwing _any_ of them away after this. Consistent with  @chevdor, the **short, fat cable** did the trick, which would seem to suggest that for some reason the tiny voltage drop across a longer or thinner cable makes the signal from the CP2102 just too weak for the chipset on a mac to interpret, which is an absurd explanation. I suppose I could pull out the old oscilloscope to validate that but....

Anyways, I had to steal the cable from my wife so I'm ordering this one (https://www.amazon.com/gp/product/B013G4EAEI) off amazon. If it doesn't work I'm sure I'll be back with some oscilloscope tracings.

Happy cable hunting!

sleepdefic1t, 2017-09-09:
Same issues.
Ended up here.

Tried better cable.
Showed up right away!

Thank you, all!

jacobtarter, 2017-09-11:
What kind of cables are you guys using that are working? I'm having this issue right now, and the ~18 anker powerline cable (along with 3 or 4 others) are all not showing the device. Seems ridiculous that it could be this sensitive to which cable is needed

sleepdefic1t, 2017-09-11:
@jacobtarter 
Well, depending on the issue, it might not be your cable.
After all the troubleshooting steps I followed, the cable ended up being the issue for me.

And it's because not all cables are equal.
Some are cheap and good, some are cheap and bad;
Some are pricey and good, some are pricey and shit.
Some only do power, some do data but poorly.

I can tell you with certainty that this is by FAAAAAAAR not the only tech that's picky about which cable you use.

chevdor, 2017-09-14:
In the first place you should avoid using Anker cables made for CHARGING.
There is no reason why charging cables would be good with data
(unfortunately). Just buy 1 good quality cable that is also not too long.

Good luck. Those issues are annoying...

jacobtarter, 2017-09-18:
The cable I was using wasn't a power only cable, although anker does make those (power line + I think theyre called).

Anyway, I hooked it up to an older desktop running elementary os and it detected it immediately.

VivienGiraud, 2017-09-27:
Thanks @fadushin !! You opened my eyes too ! I have an esp32 by Wemos and it's the same chip as you ....
Doing this and restart did it:
```
brew tap mengbo/ch340g-ch34g-ch34x-mac-os-x-driver https://github.com/mengbo/ch340g-ch34g-ch34x-mac-os-x-driver
brew cask install wch-ch34x-usb-serial-driver
```

Basirk, 2017-11-02:
So to be clear, NodeMCU ESP12E devkit boards come with different Serial chips.
Some have a SiLabs CP210x.
Some have a WCH (Winchiphead) CH340.

You can tell which you have by plugging the board into your Mac and then opening System Information. Go to Hardware -> USB. Find the There'll be a USB2.0-Serial device.
A CH340 will have Vendor ID: 0x1a86 and Product ID: 0x7523.
SiLabs website says their Vendor ID is 0x10C4 and the Product ID is 0xEA60,  0xEA70 or 0xEA71.

The latest Mac drivers for the CH340 can be found here direct from WCH (version 1.4 released Jan 2017):
http://www.wch.cn/download/CH341SER_MAC_ZIP.html
The works with my MacOSX Sierra 10.12.6.

Most documentation seems to assume you have a SiLabs CP210x chip, so if you end up here because your serial driver isn't working (like I did), hopefully you'll find this useful.

zhaxidele, 2017-12-19:
Thank you guys, solved this problem by reinstalling the driver.

jimprince, 2018-01-12:
Checking back in. I still same problem with a new cable and reinstalled driver. It's intermittent so I suppose I'll just live with it...

joedeveloper, 2018-01-29:
I am also having issues with this, have multiple boards, just one was presenting errors, tried different cables. Ultimately I decided to try a different USB port on my macbook pro - It worked!

ggrelet, 2018-01-31:
@Basirk thanks for your useful piece of information!
My USB section on the Hardware macOS pref panel says I have a `CP2102 USB to UART Bridge Controller`. But the line is shown whether my chip is plugged or not... Does this mean it is the type of my board or simply one of the countless drivers I tried to install to make my board work?

Let me precise that it's responding through the `/dev/cu.SLAB_USBtoUART` because I was able to flash a binary thanks to [nodemcu-flasher](https://github.com/nodemcu/nodemcu-flasher). The  wifi is showing under the name `ESB_1BA06D` and seems to be connectable. But the following command only shows:
```
$ ls /dev/cu.*
/dev/cu.Bluetooth-Incoming-Port /dev/cu.SLAB_USBtoUART /dev/cu.UEBOOM2-LWACP /dev/cu.UEBOOM2-LWACP-1
```

Which won't let me use the board through the Arduino official IDE.

I tried two cable made for data sharing (Android device official cable) and another one that reportedly worked on my friends MBA with the same chip (I've a MBP)

ericssonanders, 2018-02-02:
For what it is worth I will add some information regarding this issue.

Short story nodeMCU 1.0 devkit on mpb from 2013. Could not see any drivers in /dev folder. Only did see bluetooth entry in /dev/ from esplorer.

So, the drivers from silabs was the problem. When you install something on high sierra (and before?) there is a security measure that won't allow unsigned drivers to install. Looking into it i found that

```
MBP:Extensions user$ sudo kextutil -tn /Library/Extensions/SiLabsUSBDriver.kext
Password:
Kext rejected due to system policy: <OSKext 0x7ffbc27082c0 [0x7fff9d917980]> { URL = "file:///Library/StagedExtensions/Library/Extensions/SiLabsUSBDriver.kext/", ID = "com.silabs.driver.CP210xVCPDriver" }
Kext rejected due to system policy: <OSKext 0x7ffbc27082c0 [0x7fff9d917980]> { URL = "file:///Library/StagedExtensions/Library/Extensions/SiLabsUSBDriver.kext/", ID = "com.silabs.driver.CP210xVCPDriver" }
Diagnostics for /Library/Extensions/SiLabsUSBDriver.kext:
Warnings: 
    Dependency lacks appropriate value for OSBundleRequired and may not be available during early boot: 
        com.apple.iokit.IOSerialFamily - Safe Boot
```

I did not see any error when running the installer from silabs. But if you go to

System preferences -> Security & Drivers

there will be some information at the bottom of the screen regarding this. So, what I did was that I clicked on the link that is supplied in Security & Drivers to allow the driver to be installed (or if it is the installer that is allowed to run). THEN reinstalled the silabs driver but using the legacy drivers. Now, I had seen somewhere that you might had to run the installer twice. So I did that and that was it. Communication to the nodeMCU did work from inside esplorer.

[image]

pacificsky, 2018-05-18:
@ericssonanders comment is spot-on. I'm on High-Sierra and after manually attempting a kext load via `sudo kextutil -tn /Library/Extensions/SiLabsUSBDriver.kext` I got the warning prompt in Security & Privacy in Systems Preferences. Allowing the driver load there fixed the issue and /dev/tty.SLAB_USBtoUART shows up. 

Here's the note from Apple describing why only some users are hitting this - if you had the driver installed from before upgrading to High Sierra it would have continued to work transparently for you
https://developer.apple.com/library/content/technotes/tn2459/_index.html

radimek, 2018-06-07:
Hello, I would like to share my experience. I went trought all recommendations no success. The result is wrong USB-C to USB 3.0 reduction. Interesting is that USB-C reduction is working for external HDD. Strange behaviour, do you have same experience?

haukeg, 2019-02-07:
If you are having trouble on macOS...

I wrestled with this issue for a couple days on and off.  I finally fixed my issue, so I figured recounting my steps for all the other unfortunate sufferers would be worthwhile (even tho I am sure you have read 100 versions of solutions by now).  

First off, I followed most of the general recommendations and installed this [SiLabs Driver](https://www.silabs.com/products/development-tools/software/usb-to-uart-bridge-vcp-drivers) - no luck.  Then installed this [CH34 driver](http://www.wch.cn/download/CH341SER_MAC_ZIP.html) - no luck (I didn't check to determine which chip I had, just kept trying solutions).  When installing the driver(s), I saw the Security many have described, warning for a 3rd party developer trying to access some settings.  I did what most suggested and went to  System Preferences ➜ Security & Privacy ➜ General and made sure my allowing apps to be downloaded from "Mac App Store and identified developers" - which it was.  Still - no luck.  Then, after hitting issue after issue, I resorted to assuming I had bad Micro USB cables (charge only) - so I ordered a new USB Sync cable to remove that variable.  

I could never get my ESP8266 NodeMCU LUA CP2102 ESP-12E (made by HiLetgo) to show up under Tools > Port in the Arduino IDE (should see "/dev/cu.SLAB_USBtoUART"). 

After banging my head on it over and over, I tried re-installing the CH34 driver - which popped the security warning again (as described above). This time I clicked the "Goto Settings" button - which simply took me back tot the same System Prefs Security dialog I checked for the "allow apps" setting before.  **However, this time I noticed there was a subtle macOS option next to those "allow" settings which said "Some system software was blocked from loading."** and here was an "Allow" button.  When I tapped the "Allow" button, it popped a dialog that showed both the SiLabs and Ch34 drivers unchecked, which I fixed.  After that, voila!  It all worked (and on all my USB cables no less).

I cannot say which driver was my solution, but thought the important thing to point out to anyone struggling was to look for the Allow button in Security - I didnt notice it because all solutions I read just called out the "Mac App Store and identified developers" setting in that panel.  I had that set before installing the driver and it DID NOT stop the drivers from being blocked, so check for that.  Hopefully this helps someone else, what a pain...

hoosierEE, 2019-02-08:
Thank @haukeg for the new information.  Do you happen to know the type of Mac used?  I've seen slightly different problems between the 2016+ Macbook Pro laptops "with touchbar" versus "without", even when they're supposedly running the same version of MacOS.

mrsufgi, 2019-03-12:
@haukeg had exactly the same problem, same solution. Allow your driver to run! 

MBR 15 touchbar 10.14.1

existeundelta, 2019-04-14:
tried several times , normal, legacy cp2101, ch34, alert in security don't show, also disabled spctl kext-consent, tried several cables
 @haukeg very similar as you... but did't see:

> "However, this time I noticed there was a subtle macOS option next to those "Allow apps downloaded from" section which said "Some system software was blocked from loading." and here was an "Allow" button"

jnettome, 2019-09-05:
GUYS, REMEMBER TO TURN ON YOUR BOARD

😱 🤯 😭