nodemcu/nodemcu-devkit-v1.0 #3: Cannot flash the firmware from a Mac

marcoskirsch, 2015-07-08:
I have OS X Yosemite. SiLabs drivers for the USB-Serial adapter are installed and working. I can communicate to the board via /dev/cu.SLAB_USBtoUART and I can use it in screen, ESPlorer, esptool.py... but strangely not minicom.

However, I haven't been able to flash new firmware successfully.

Can anyone confirm this can be done? I did it on older ESP01 hardware before but I don't rule out me doing something stupid. 

* What tool did you use to flash?
* What firmware image?

I posted additional information here: http://www.esp8266.com/viewtopic.php?f=22&t=3903 but nobody has been able to help. Help is appreciated!

abl, 2015-07-08:
Minicom requires some tweaks to work correctly, iirc - it has to be told to ignore certain signals and lines that are being repurposed.

Ok, so you are actually flashing the build and getting to boot mode 3, which is boot from SPI flash. (Esptool, with a supported USB->Serial, is able to reset devices and put them in to boot mode 1, which is what you use for writing to the flash.) Secondary boot mode not being 7 I tend to associate with power supply issues. I've never found a concrete answer as to what it means.

Based on other bug reports, it looks like you might have a defective flash chip or a nonstandard flash chip - or you're flashing bad firmware. Have you tried flashing the stock NodeMCU firmware? Do you have access to a Windows machine to try the official NodeMCU flasher?

marcoskirsch, 2015-07-08:
I doubt but don't completely rule out a defective flash chip because it was functional out of the box. I do have access to Windows PC and will try that next. In order to rule out a bad firmware image, can you link to an image that is known to work on NodeMCU devkit 1.0?

abl, 2015-07-08:
http://www.esp8266.com/viewtopic.php?p=15306 kind of indicates that, based on your error, something weird is going on with your flash chip. It might need a specialized flasher or a different header. This wouldn't have been a problem...until you tried to flash it or write a large amount of data to flash.

The NodeMCU flasher comes with NodeMCU firmware. I'm not sure which version, though.

Official NodeMCU releases are at https://github.com/nodemcu/nodemcu-firmware/releases - I don't have a NodeMCU 1.0 to test on but I'd expect a lot of issues filed if any of these builds were broken.