nodemcu/nodemcu-devkit-v1.0 #4: V1/V2/V3 difference ?

asetyde, 2015-08-18:
Schematics say about v 0,9 .. but many type at now .. v1 v2 v3 ... therefore ? thx

mqu, 2015-10-25:
you could have a look here : http://frightanic.com/iot/comparison-of-esp8266-nodemcu-development-boards/

asetyde, 2015-10-25:
Many Thx , 

I also understand with my hand this difference , it's a good choice model v2 also doit v3.

In particular about power , use reserve pin to give a special port is more safe and practical that use vin/gnd as bridge to give power.

Minimum power is about 500ma , after 6 month of use I say 1000ma is good .
With sensor and rele 1.5k ma or 2k ma because this board is sensible to lower power down , strange error may appear if power is not enough to cpu minimum needed .

But this board , I think , is a big big successful ! This community also too.

vowstar, 2015-11-24:
NodeMCU only released V0.9 and V1.0 board. The next is V1.1, we plan to release it in 2016.
The main difference is USB to UART bridge IC. V0.9 uses ch340G, but have many problems on MAC, so we released V1.0 which uses CP2102 and have a good performance.

odoveiz, 2016-03-25:
Any news on a new version?

asetyde, 2016-03-29:
on closed issue you can find info , use search

asetyde, 2016-03-29:
https://github.com/esp8266/Arduino/issues/1481

asetyde, 2016-03-29:
only nodemcu v2 as nodemcu 1.0 is amica / nodemcu compliance 
( v3 has usb different chip ch340 and some power difference / v1 is nodemcu 0.9 )

i suggest use nodemcu 1.0 amica ( v2 ) see post how check it

closed 2016-03-29 by asetyde

jk2552, 2017-10-05:
hi folks i need Datasheet of Nodemcu ver0.1 kindly anyone help me..

marcelstoer, 2017-12-27:
Sorry to comment on an old and closed issue. My post that was mentioned above has a photo at https://frightanic.com/iot/comparison-of-esp8266-nodemcu-development-boards/#official-vs-unofficial which shows in detail how an "official" 1.0/v2 devkit has to look like.

To my knowledge neither Banggood nor Aliexpress still sells the Amica v2 devkit. Seeed Studio still has them but it's marked as "out of stock". If you [search taobao.com for "NodeMCU cp2102"](https://s.taobao.com/search?q=NodeMCU+cp2102) it turns up quite a few that look genuine.