nodemcu/nodemcu-devkit-v1.0 #5: Unable to connect to dev kit at all

zdrc, 2015-09-15:
Hi, I recently purchased a nodeMCU dev kit off of amazon.

I'm running Mac 10.10, and I have the most recent CP2102 driver. I own a couple of CP2102 breakouts that I can connect to no problem. 

I am entirely unable to get any kind of response out of the dev kit using the micro USB port. Nothing shows up under /dev/tty.* and IO Registry Explorer is unable to locate the device in the USB port. I've tried to circumvent the CP2102 chip on the dev kit using one of my breakouts (connecting txd and rxd) and I seem to be able to connect to the module. I was having trouble communicating with the firmware (it was acting a little erratic) so I decided to reflash the AT firmware then the nodeMCU firmware. After reflashing I am unable to get any kind of response out of the chip over serial (through either AT or nodeMCU).

Is there anything I can do or is the module simply defective?

vowstar, 2015-11-24:
Please check
/Library/Extensions/SiLabsUSBDriver.kext
$ ls -lah /dev/cu*
may be it is not /dev/tty*

zdrc, 2015-11-24:
I have the driver. That directory checks out. It's not in /dev/cu*.

This is old, but I now believe me problem was I set the flash setting improperly. Unfortunately, I don't have access to the dev kit anymore.