nodemcu/nodemcu-devkit-v1.0 #9: Vin 5V not working

confile, 2015-11-11:
I the pin image you labeled VIN 5V. This does not work by default. Is there a way to activate it?

vowstar, 2015-11-24:
Between Vin 5V and VDDUSB is a diode, 1n5817, Please connect USB cable and test the Vin voltage, and test LM1117's Vout. And then disconnect USB cable, supply Vin 5V to 9V, redo the test. If no voltage, please check the wire between Vin and LM1117

confile, 2015-11-25:
I connected the dev kit to USB. In this case Vin 5V has no power. What is Win 5V used for. Is it used for an external power supply instead of USB? Does it mean you can plug the dev board to a 5V power supply instead of USB? Is Vin 5V only input power but not output power, i.e., when you use USB to power the dev board there is no outgoing power of 5V on Vin 5v?

PaulHam211, 2016-01-24:
Is VIN only for input power? I have a pref connected to VIN and its not powering.
Shame because i need to have a external 5v PSU while prototyping, I should be able to tap 5v from the USB.

asetyde, 2016-01-25:
also output 5v power

christianbur, 2016-02-14:
is it possible to connect a 5v sensor to vin 5v (the power supply is usb)?

asetyde, 2016-02-15:
Yes , i use it

remedix, 2016-02-28:
@vowstar so vin takes 5V to 9V ? So if I connect a 9V battery to 'VIN' pin and any 'G' pin that would power it up right ?

MrAda, 2016-08-12:
I have tried to plug USB cable into board and then did a reading on all of the power pins.  The three 3.3V and the one 5.0V.  Only the 3.3V seem to be providing power.  The 5.0V has really low voltage, not 0 but really low.  @asetyde stated he uses it.  Maybe he can tell us what we have to do to make it happen?  I want to be able to power the board with a 5V source.

shadowhorst, 2016-08-13:
I had the same issue with my VIN pin not providing any voltage. But in my case there is also an VU pin on the board which gives me 5V output voltage.

MrAda, 2016-08-13:
I got lucky and mine works.  You can't plug the USB cable in while powering trying to power with the 5V pin.  I have to see where this VU pin is.  I just checked my PNG against my board and the PNG is wrong!  It doesn't show VU pin.  Do you have a good picture I can see that describes the pins?  This is great news to me as I was trying to power a relay and couldn't with the 3.3V output.

MrAda, 2016-08-13:
Grrrrrrr.  My VU pin is showing less than 1V.  I really need to use this for my relay.

MrAda, 2016-08-13:
You have to have usb plugged in

drawkula, 2016-08-13:
[This NodeMCU boards comparison](http://frightanic.com/iot/comparison-of-esp8266-nodemcu-development-boards/) shows the pinouts of the commin NodeMCU siblings.

VU (USB power) is an invention of the ptoducer "LoLin" and the other siblings don't have it.

MrAda, 2016-08-13:
I am so happy that I picked the right sibling!  My Version 3 LoLin has this and now I can control my relays.

shadowhorst, 2016-08-14:
@drawkula Thanks for the great link. Have been looking for exactly something like that when I was choosing a NodeMCU board.

supersjimmie, 2016-08-31:
My NodeMCU board does have this VIN pin (not the VU).
When I only power this VIN with 5V, the board does not boot. (short blink of blue led, then nothing)
When I also connect USB to my laptop it boots fine and keeps running on the VIN 5V when I then disconnect USB.

Tried it with 3 different 5V sources, an unknown 5V/1A adapter and 2 Apple adapters (1A and 2A versions). I did this with a stripped USB cable to extract the 5V directly from the adapter and inject it into the VIN 5V pin.
(so to be clear, I don't connect the adapter to the USB but to the VIN)

I don't understand why the board only needs USB power to boot and then runs fine without it.

MrAda, 2016-08-31:
Do you know which version of the board you have?  Sounds like you don't have the latest.  The VU pin, third down on the left side (next to ESP chip) has  a 5V out pin called VU (or VV maybe my eyes can't see it).  I have tried to power this unit with wires instead of the usb and I don't remember if I was successful, or I just couldn't tell.  Need the usb to download to it.  If you downloaded a lua file that just blinked the led then you could tell if the power sources work or not as a running app.  Eventually I'll have to pull power off of a brick or buck converter so once I get some new ones in, I'll report how well it goes.

supersjimmie, 2016-09-02:
My NodeMCU is v2. I have now discovered that, probable because there is a protecting diode that lowers the voltage just a bit, my module does boot but fails to connect wifi. As soon as I power with usb, the voltage gets just a bit higher and wifi connects.  I will try to solve this by searching a power supply with a bit more than 5V.

oshamed, 2016-09-24:
It works for me using 7.5 volt adapter to 5 volt regulator then to my esp

Imara90, 2016-10-12:
@supersjimmie did you end up solving this? I have the same problem

tablatronix, 2017-03-23:
I use VIN works fine, maybe you have voltage drop on your 5v, probably need 5.2v

StalePhish, 2018-03-11:
I have the ESP12E DEVKIT V2 board with NodeMCU (from HackerBox) and I'm experiencing the same problem.

If I plug in USB, it boots and I get 4.65V off the VIN.
If I plug in a 9V battery to VIN and GND, it does NOT boot, and I get around 8.5V off the VIN. It's like when you plug in USB, it's doing something different.

rsagittario59, 2018-03-19:
My NodeMCU board does have this VIN pin (not the VU).
When I only power this VIN with 5V, the board does not boot. (short blink of blue led, then nothing)
When I also connect USB to my laptop it boots fine and keeps running on the VIN 5V when I then disconnect USB.

Tried it with 3 different 5V sources, an unknown 5V/1A adapter and 2 Apple adapters (1A and 2A versions). I did this with a stripped USB cable to extract the 5V directly from the adapter and inject it into the VIN 5V pin.
(so to be clear, I don't connect the adapter to the USB but to the VIN)

I don't understand why the board only needs USB power to boot and then runs fine without it.

supersjimmie 
StalePhish
Imara90
did you solve this problem? if yes .... How??
many thanks

bartieman, 2018-03-24:
I discovered that for my Lolin NodeMcu v3, the VU pin is only outputting 5V (approximately) if I power the board by USB. If I power the board with 5V using VIN and G, the VU pin is only outputting around 1.1V.

tablatronix, 2018-03-24:
It never occured to me that you guys are not using actual nodemcu boards, are you all using lolin r some other knockoffs ? I have amica nodemcu v2 boards and do not have an issue.

MrAda, 2018-03-24:
I have been watching these messages.  I have successful using my Lolin board but I basically had to feed a power brick to my USB cable which plugs into the usb port on the board.  This provides power to board and the other pin didn't seem to work out well.

rsagittario59, 2018-03-26:
Many Thanks for your answer

Inviato da Posta per Windows 10

Da: Chris Sparks
Inviato: sabato 24 marzo 2018 19:29
A: nodemcu/nodemcu-devkit-v1.0
Cc: rsagittario59; Comment
Oggetto: Re: [nodemcu/nodemcu-devkit-v1.0] Vin 5V not working (#9)

I have been watching these messages. I have successful using my Lolin board but I basically had to feed a power brick to my USB cable which plugs into the usb port on the board. This provides power to board and the other pin didn't seem to work out well.

rsagittario59, 2018-03-26:
Many Thanks for your answer

Inviato da Posta per Windows 10

Da: Shawn A
Inviato: sabato 24 marzo 2018 15:14
A: nodemcu/nodemcu-devkit-v1.0
Cc: rsagittario59; Comment
Oggetto: Re: [nodemcu/nodemcu-devkit-v1.0] Vin 5V not working (#9)

It never occured to me that you guys are not using actual nodemcu boards, are you all using lolin r some other knockoffs ? I have amica nodemcu v2 boards and do not have an issue.

rsagittario59, 2018-03-26:
Many Thanks for your answer

Inviato da Posta per Windows 10

Da: bartieman
Inviato: sabato 24 marzo 2018 15:11
A: nodemcu/nodemcu-devkit-v1.0
Cc: rsagittario59; Comment
Oggetto: Re: [nodemcu/nodemcu-devkit-v1.0] Vin 5V not working (#9)

I discovered that for my Lolin NodeMcu v3, the VU pin is only outputting 5V (approximately) if I power the board by USB. If I power the board with 5V using VIN and G, the VU pin is only outputting around 1.1V.

togius, 2018-04-13:
Use VV Nodemcu port for 5V output.
[https://frightanic.com/iot/comparison-of-esp8266-nodemcu-development-boards/](url)
Best answer drawkula. thanks.

661995, 2018-05-29:
hello guys ?
i have a nodeMCU esp12E ELU devlpmnt board.. i m connecting 4 6V relay to vin pin. when i plug in a USB to my computer 1st and second relay is working fine . but when third relay is on my nodeMCU board was short. means vin and gnd was shord inide automatically and board is died.. someone plz help me to fix it.. how i solve this? can i use 6v relay instead of 5V relay ? is any connection issue in my prototyping ? san we must use a optotoupler for protection of microcontroller ? somebody plz help me