nodemcu/nodemcu-devkit-v1.0 #10: Schematic PDF file is broken

jjorsett, 2015-11-21:
Adobe Acrobat X Pro is unable to read file NODEMCU_DEVKIT_V1.0.PDF. Acrobat says, "Acrobat could not open 'NODEMCU_DEVKIT_V1.0.PDF' because it is either not a supported file type or because the file has been damaged."

vowstar, 2015-11-24:
Please re-download it or reinstall your abobe, or use chrome open it. We tested this file and no problem.

jjorsett, 2015-11-24:
I am able to view it with Chrome, but both Adobe Acrobat X Pro and Adobe Reader XI (both of which I reinstalled, just to be sure) are unable to open the file (which I downloaded a new copy of). The only reason I care is that I'd like to extract some of the images and recreate them in a more legible size. I'll just use screen capture from a zoomed Chrome window to do this instead. Thanks for trying.

closed 2015-12-25 by vowstar