nodemcu/nodemcu-devkit-v1.0 #12: Suggestions about new NodeMCU devkit based on ESP32

nodemcu, 2015-12-11:
Hi, all
since esp32 is out for beta, the chip is much more capable than esp8266.
I'd like to know that what features we should added to our new devkit.
usb chip? flash size? pin-to-pin compatible with devkit v1.0?
Thanks in advance.

TerryE, 2015-12-11:
Some things that we do know about the esp32 (compared to the esp8266:

*    Espressif have opened a new support site: [esp32.com](http://esp32.com/), but which is accessible by registered users.
*    It runs a dual core LX108 (rather than the single core LX106 on the 8266)
*    It has a lot more RAM ?128Kb or 256Kb? rather than 48Kb
*    It has more pin-outs and a richer set of peripherals.

[image]
Very little further detail is approved by Espressif for discussion in the public domain, but the dual core / single memory space aspects mean that the simple non-OS SDK is unlikely to port well.  You need some form of centralised lock management and resource allocation as is provided by RTOS, and hence any new SDK will need to run under RTOS.   
 
There is so much that we don't know and until we do it is difficult to make concrete plans.

*    Will there be a material pricing differential between the esp32 and the esp8266?
*    What will be its availability and when?
*    Can the core developers get NDA access to the device specs?
*    Will Espressif keep the two devices in manufacture or will they be soon phasing out the esp8266?

Clearly the esp32 is the future, but to me one key issue is how long we have to maintain parallel support, because our development strategy will depend on where this is months or more than a year.  Another is the availability of esp32 hardware.

asetyde, 2015-12-11:
Very interesting !!!

AdySan, 2015-12-22:
Would love to see a JST socket to connect a LiPo battery + support for USB charging. Basically something like the Adafruit Feather HUZZAH.

vowstar, 2015-12-25:
I think a bran-new board is needed, because ESP32 have much more GPIOs, DAC, touch, etc. 
The new board must very small and  could carry it in the pocket, and have on-board USB to UART.
We have found a good and cheap DC-DC power IC, more small and stable for new esp32. So I suggest that use DC-DC in new dev-board. And many user like IPEX interface and external antenna, so I think add both PCB antenna and IPEX interface is better.
The GPIO layout will looks like devkit1.0(amica) as much as possible, but can not pin-to-pin.
@AdySan I think we can make a core board and a battery board which including charger and PMIC, and battery interface, using stacked design.

AdySan, 2015-12-25:
Yeah a separate board would do. One thing incenvenient about NodeMCU dev kit is that the deep sleep mode is rather pointless since the USB-serial IC drains way more current even when the ESP8266 is in deep sleep. It would be nice to have the capability to un-power the USB-serial chip with a jumper or a 0 Ohm resistor.

sathishvasimalai, 2016-02-25:
how to buy esp32 development kit

goldenbr0wn, 2016-05-30:
More analog input would be great ! Maybe using a multiplexer ?

pathob, 2016-09-04:
Please, do not use a different pin mapping again (if it is not too late already). This can be really confusing...

jeffeb3, 2016-10-31:
Dunno if anyone is still looking at this, or if it will matter, since you might be well on your way to designing this board since the 32's have been sort of available already, but here are some of my thoughts:
1) There is a lot more I/O, and it will be hard to tell what people want. Maybe you could have a sort of break away section with a ton of I/O, but you can cut it off if you really just want some of the new features with a similar footprint to the current nodemcu.
2) The main focus of a subset of pins would be PWM, ADC, SPI, I2C, UART. I think you should be easily be able to get 12 or so pins and cover those things pretty well. A larger board version, or a version with a breakaway section can have an emphasis on the other port types.
3) I never messed with the sleep stuff of the nodemcu0or1, but from what I read you need to connect GPIO16 to EN. Maybe just do that in a trace or through a 0 Ohm resistor, and save us the space by not giving us GPIO16 or EN.

It really seems like you kind of want something like a mini and a mega version. The mini for battery/sleep/small stuff and the mega for easy/everything/touch. Ideally, the differences would be small enough that you could always run the same code on each, so they would need the same flash and general setup, you just couldn't connect the stuff that needed extra IO to the mini version. Maybe you also couldn't run off of a battery for the mega version, or some kind of compromise like that.