nodemcu/nodemcu-devkit-v1.0 #14: Impossible to flash/communicate

RataDP, 2016-02-02:
I'm having troubles to flash or connect my _devkit v1_. It arrived my home 2 days ago and I was not able to flash it or at least communicate with it.

I'm going to describe my case. I connect my _devkit v1_ via USB to microUSB with the CP2102 USBtoUARP, then the blue led of the ESP8266 blinks twice and stop. After this I can see the SSID "AI_THINKER_DA7E19" and connect to it, I got a 192.168.4.2 and the device .1. It answers to the ping but there is not webserver. Now I try to flash it, first I tried with Linux and ESPTool,

``` 
# python2 esptool.py --port /dev/ttyUSB0 write_flash 0x00000 firmware.bin
Connecting...

A fatal error occurred: Failed to connect to ESP8266
```

Well I read somewhere it could be a bug with ESPtool with high values of baudrate, so I tried with `--baud 9600` but the same output also with different baudrates. Meanwhile the program is in the state `connecting...`, the blue led blinks. I tried to reset with the Flash button pressed and the same result, nothing change. Also I tried wiring the GPIO0 to GND but still the same error.

Now it is the turn for Windows, I tried with the nodemcu-flasher, the driver was correctly install and the port is assigned correctly too. Select the COM port and set the baudrate to 9600 and flash. The log says `Begin to find ESP8266` enters in a loop and the blue led blinks every 1/2 seconds. Just to point it out, in the tab *operation* not appears any mac, it says "Waiting...". Just like before I changed the Baudrate and I tried all of them but nothing happens. _(I also press flash and reset like before but nothing)_

After all this troubles I tried to communicate with the devkit via Serial port with putty and Screen. The connection seems to be established but nothing appears in it. Pressing RST it should write something... garbage or some characters but nothing.

I don't know if I'm doing something wrong or the devkit it's broken. Some solution or tip?

Thank you

PD: I posted it in the old devkit repository, now it is in the correct repository.

EDIT:
@mrmonteith answered in the other repository.
``` 
I'm trying to remember when I first tried this. I think I used the flasher tool. You start the attempt at transfer, hold down the reset and flasher button, release the flasher button and then about a half second or so after release the reset. I don't have my device in front of me to check. Of course I only had to do it one time and I've had mine for about month. Hope it helps anyway. 
```

I tried too, but no works it.

prob1995, 2016-10-04:
Same issue here... Still can't flash the firmware...

gurubhat, 2016-10-21:
change the USB cable.. had the same issue, resolved after changing the cable.

KOKONUTCREMEOG, 2017-05-29:
And here same issue. I operate PC Windows 10, And Android...,.But it's waiting for mac

ChrisHeerschap, 2017-06-22:
Found this page when dealing with the same problem. Swapped out USB cables to no avail. A little bit more searching and I found this which wound up fixing it for me - maybe it might be helpful to others:

http://kig.re/2014/12/31/how-to-use-arduino-nano-mini-pro-with-CH340G-on-mac-osx-yosemite.html

JayPalm, 2018-06-19:
Hey, did you ever figure this one out? I'm having pretty much the same issue on a Mac with a nodeMCU Amica board (uses a CP102 USB-UART bridge, installed the SiLab driver, etc). I have tried multiple baud rates but always get the same response: A fatal error occurred: Failed to connect to Espressif device: Timed out waiting for packet header.

I have tried multiple USB cables and multiple NodeMCUs to no avail. Really bummed right now...

ChrisHeerschap, 2018-06-19:
@JayPalm did you try the link I listed in [my last comment](https://github.com/nodemcu/nodemcu-devkit-v1.0/issues/14#issuecomment-310425447)?

Parin810, 2018-06-21:
Facing the same issue.. Anyone found the cause of the problem? or a solution that worked..

mrmonteith, 2018-06-29:
Any luck yet?  I had a couple of thoughts.  Mostly around possibly weak components.  Do you have any sort of oscilloscope? I bought a Hantek digital one that connects to the computer and does fairly good.  That would go a long ways to checking things like power supply voltages and noise on the lines.   But without you can still probably locate the issue.   A couple of things you can try.  That is put a capacitor between 3v3 and Gnd as an extra filter.  USB interfaces are notoriously noisy.    There is also an Espressif forum for issues with the ESP8266 and ESP32 devices.  You might post there too.

VinayKadalagi, 2018-11-25:
Just reconnect the usb cable keeping the flasher open 
Look for log for below line ->
> Note:Detect serial port changed.
> Note:Auto MAP serial port.Port-->COM5
Then you should be able to flash. Looks like the tool cannot detect already connected device.

ersyncar, 2019-03-28:
i had same issue. i change usb cable and tried different values for advanced setting on Nodemcu. and it worked by 
-Baudrate 9600, -flash size 4Mbyte, -flash speed 26.7mhz, -spi mode DIO

ThreeSpy712, 2019-03-29:
Try holding down the GPIO/reset button on your MCU, and then flash whilst still holding down. This worked for me.

Joshverge, 2019-03-30:
Same issue. Won't flash

Joshverge, 2019-03-30:
> Just reconnect the usb cable keeping the flasher open
> Look for log for below line ->
> 
> > Note:Detect serial port changed.
> > Note:Auto MAP serial port.Port-->COM5
> 
> Then you should be able to flash. Looks like the tool cannot detect already connected device.

> Just reconnect the usb cable keeping the flasher open
> Look for log for below line ->
> 
> > Note:Detect serial port changed.
> > Note:Auto MAP serial port.Port-->COM5
> 
> Then you should be able to flash. Looks like the tool cannot detect already connected device.

That's not the issue, it's able to detect the right com port right from the start

ersyncar, 2019-03-30:
> 
> That's not the issue, it's able to detect the right com port right from the start

did you try by another usb cable. i had same issue and changed the cable. i tried by my printers cable and it worked :) 

and i attached the picture of working settings on my ESP8266 -ESP 01

[image]
[image]
[image]

Joshverge, 2019-03-30:
Okay I resolved this for me. You just need to hold down the flash button until you the status bar moving forward after pressing flash

horodchukanton, 2020-08-10:
Mine was fixed by putting the board to an oven for two minutes. Looks like a weak components issue

Blazer9419, 2023-01-26:
I am also having the same issue. It says “waiting mac” however my FTDI and ESP is flashing red and the ESP was also flashing blue for a couple seconds then stopped. Is there anything I am doing wrong here?
[image]
[image]

TSRider85, 2023-11-25:
I had the same problem but i solved it, it was my usb port. Looks like it the current of that one was to low. So try a different usb port!