nodemcu/nodemcu-devkit-v1.0 #15: Maximum Vin

adrianlzt, 2016-02-29:
I want to use the Vin pin to power NodeMCU, but it's not clear which is the maximum value we can use.

In the rear of the PCB it's written "Vin +5v recommended, +10V MAX".

But in this [pdf](https://github.com/nodemcu/nodemcu-devkit-v1.0/blob/master/NODEMCU_DEVKIT_V1.0.PDF), in the "POWER" section is written:
"Max Supply Voltage: 20V"

adrianlzt, 2016-02-29:
Looking at the [datasheet](https://github.com/nodemcu/nodemcu-devkit-v1.0/blob/master/Documents/NCP1117ST33T3G.PDF) of the power regulator

Vin should be between 4.75V and 10V

closed 2016-02-29 by adrianlzt

miguelpom, 2017-01-23:
Hi @adrianlzt.
I'm working on a project with the NodeMCU v1.0. Together with the Node I have an Arduino Nano running. Thus, to simplify my hardware design, I want to power the NodeMCU from a 5Volt external power source (the same that powers the Nano board).
Since you closed this issue, did you tried to power the NodeMCU from its Vin pin?
Were there any performance issues? 
What's the version of the NodeMCU board you used?

Thank you in advance!

adrianlzt, 2017-01-23:
I was just exploring how can I power the NodeMCU, but haven't done it yet, sorry.

nedimhx, 2017-02-28:
Hello everyone, 
@adrianlzt @miguelpom I'm working with NodeMCU for over 5 month, and it's time to test it on Maxell 9V alkaline battery. From 5 to 10 Vin is perfect solution.

downlz, 2017-09-14:
Thanks for the information.I was looking for the input voltage.The board pins more or less matches arduino Nano.

zfarrugia, 2017-12-03:
I can confirm that 15.5V will kill the board ... from experience.

mbfac, 2018-04-06:
Very bad experience with 12V.... works fine 1 minute.... :-(

Jmilesy, 2018-04-28:
However, after that 1 minute at 12v, you have a lovely bench heater, again from experience!!!

mmaxus35, 2018-08-14:
Thanks for the comments guys. I thought that i can handle up to 18V .. Yet, burned out at 16v :(

shakthi-arul, 2019-02-13:
This thread is useful. Thanks, guys!

fryguy04, 2019-02-26:
I had a NodeMCU in my attic working just fine for 6 months on 5V ... replaced it with 12V and ever since then it has been cyclic rebooting, it will work fine for a few minutes (maybe 5min) then reset itself. Converted back to 5V and back to working fine again ... Thanks to this Thread! I too thought i could use upwards of 12v, seems not to be true

gerritfries, 2019-07-12:
I have a NodeMCU Amica powered with 16V (the voltage of my doorbell) and it's working fine for two months now. I guess even though the voltage is a bit too high the ampere might be pretty low and thats the reason it works without problems.

joao-carvalheira, 2019-09-27:
> Hello everyone,
> @adrianlzt @miguelpom I'm working with NodeMCU for over 5 month, and it's time to test it on Maxell 9V alkaline battery. From 5 to 10 Vin is perfect solution.

Did it work?

Soraxem, 2020-07-07:
Mine turned to a heater at 8Volt?

L4RGER, 2020-10-28:
Im having trouble with a 5v power supply in at Vin. With power applied to board doesn’t work until reset button is pushed and then it works fine until next power cycle. Then it requires a reset again. 
Going to rewire and supply 9v to Vin

ROBBED666, 2020-12-09:
My NodeMCUs have AMS1117 3.3 onboard step-down regulator which has a Max Vin of 15 Volts DC (taken from the device spec sheet). I've been bench running a unit at 14 Volts for 24+ hours with no excessive temperature rise, or operational issues.  I in about 50mA, therefore Power dissipation of about 0.6 Watts in the regulator as it's a linear device.

For use in a vehicle where battery charging can push, or even exceed 15 Volts from the vehicle's alternator, this is a 'little close to the wind'. A quick and dirty solution is to stick a couple of rectifier diodes in the supply lead (each will give approx 0.6Volt drop each. Might want to add a reservoir capacitor after the diodes just to smooth things out. (100uF or so.)

Disclaimer:  Use the above info at your own risk, yada, yada. If you don't understand it, don't do it!

hariharan69, 2021-02-26:
> I want to use the Vin pin to power NodeMCU, but it's not clear which is the maximum value we can use.
>
> In the rear of the PCB it's written "Vin +5v recommended, +10V MAX".
>
> But in this [...]

boards like Arduino and nodemcu all share common voltage regulator called AMS1117 that can support upto 18V its specifically optimized to run in 12v lead acid voltage ranges ie: it cuts of at 10v(min voltage) so go ham and plug a car battery to it :)

CaptainJack42, 2021-04-26:
My experience shows that powering it with ~10.6V will kill the board (after like 15 Minutes of continuous operation) after that the board is only good for using it as a pocket warmer that can be powered by usb or to light up your room using the red and blue onboard LEDs

nilesh-akhade, 2022-09-17:
Sorry, I'm late for the party. But please make it simple for me.

But still, how do I power my NodeMCU 1.0 (ESP-12 E Module) / [LoLin NodeMCU](https://www.make-it.ca/nodemcu-details-specifications/). What are current and voltage requirements? Are there any power modules available?

From my research, I have options:

Voltage: 4.5V-10V or 5V or 3.3V
Current: 500mA or 1A

mbfac, 2022-09-17:
Hi

Vin pin between 4,5 and 10 V MAX  but  5V is better... more safe!

Best regards

nilesh-akhade, 2022-09-17:
That is very helpful @mbfac. The conclusion is
- Vin pin expects anything between 4V to 10V,  500mA - 1A. (5V recommended)
- MicroUSB input expects: 5V, 500mA

RichyT, 2022-10-23:
The regulator on the board is a linear regulator. It has to dissipate heat equal to the current multiplied by the voltage difference. For 5V->3.3V, that's not a lot. For 15V->3.3V, that's going to be a whole lot more (almost seven times as much). It's only a small device with no heatsink.

If you're wanting to run off 12+V and you're doing anything that draws any power, you might want to consider using a beefier regulator to drop the voltage first (15->5 for example) or replace the regulator with something beefier or non-linear like a buck converter.

RichyT, 2022-10-24:
> Looking at the [datasheet](https://github.com/nodemcu/nodemcu-devkit-v1.0/blob/master/Documents/NCP1117ST33T3G.PDF) of the power regulator
> 
> Vin should be between 4.75V and 10V

Just to say, I had a look at that datasheet and I think you're misinterpreting it. Though it does seem anecdotally that 10V is the limit.