nodemcu/nodemcu-devkit-v1.0 #16: digitalWrite HIGH & LOW inverted

MutaborLingua, 2016-05-09:
Hello!

Let me get this straight. I bought the following board:
http://www.aliexpress.com/item/Free-shipping-NodeMCU-development-board-for-ESP-12E-from-ESP8266-esp-12E-Lua-IoT-programable-wifi/32583355412.html

I use the Arduino IDE, and everything works pretty much fine. I only have one problem:
If i test it for the blink sketch it reads HIGH as LOW and LOW as HIGH.

Example:
[code]
void setup() {
  // initialize digital pin 16 as an output.
  pinMode(16, OUTPUT);
}

// the loop function runs over and over again forever
void loop() {
  digitalWrite(16, HIGH);   // turn the LED on (HIGH is the voltage level)
  delay(1000);              // wait for a second
  digitalWrite(16, LOW);    // turn the LED off by making the voltage LOW
  delay(100);              // wait for 1/10 second
}
[/code]

This should do (as far as i know) that the LED is on for 1 second and then have a short pause of 1/10 second.
But what the board is really making is the opposite. The LED is off for 1 second following a short blink 1/10 second and so on...
Am I insane or is this plain inverted?

I'm on a mac with Arduino IDE 1.6.8
I use the following bridge drivers:
https://www.silabs.com/products/mcu/Pages/USBtoUARTBridgeVCPDrivers.aspx

Thank you for answers!

pamribeirox, 2016-06-19:
Some of those boards have the LED connected to VCC, probably because the MCU can sink more /the needed) current in the I/O pins at lower level.
This way, the LED only lights in the LOW state of the pin when it outputs about 0V

tabvn, 2016-07-02:
I got the same issue , set LOW led is on. 

and  digitalWrite(relayPin, HIGH); // Led OFF i tested both on D0// 16 , and D7 // 13

very confusing about this ?

pathob, 2016-09-04:
This is the wanted behaviour. If the LED would need a current of 20 mA for full brightness and you would connect the anode of the LED to the GPIO, the GPIO couldn't provide the needed 20 mA, as it can only provide 12 mA.

If you turn it vice versa and connect the cathode to the GPIO and the anode to the main 3.3 volts power supply (which should be able to provide about 800 mA of current), you can reach the full brightness (and save your GPIO). But then, the LED lights up, when the GPIO is set low.

tabvn, 2016-09-05:
Thanks for your information that helped. @pathob

speeddragon, 2017-01-24:
Just to be clear, is normal for every Digital PIN do be inverted or just the pin LEDs ?

Trafitto, 2017-07-17:
Same problem with digitalWrite(ledPin, 0); //Turn led ON

mharmon12, 2017-08-14:
Very common practice to do this.
An I/O typically sinks more current than it can source.

Find a schematic for the LED on the circuit board you are using and you'll see how it's biased for the I/O.
Writing a "1" to that port doesn't mean turn the LED on, it means write a "1" to that port.

If this is difficult for you to remember then there are ways around the confusion.
I use sharp defines in my code to create pseudonyms to add clarity.

Someone once said that good code reads like a story.

Coding in C is not an elegant story...
The following code looks more like telling your pet to do a trick.
I address my pet (he knows I'm talking to him) then the trick I want the pet to do.

// My pet's name
#define Red_LED 16

// My pets mode
pinMode(Red_LED, OUTPUT);

// Your pet's tricks
#define turn_On 0
#define turn_Off 1

digitalWrite(Red_LED, turn_On);
digitalWrite(Red_LED, turn_Off);

You could get really crazy and write a function...

void RED_LED( string action ){
    if( action == "ON" )
        digitalWrite(Red_LED, turn_On);
}else{
        digitalWrite(Red_LED, turn_Off);
}

RED_LED("ON");
RED_LED("OFF");

I would dig into the pins_arduino.h file and see if there is a definition for the LED on that GPIO and use that reference. It's a good practice to do that because it makes your code more portable.

NarinLab, 2017-09-15:
How can i make my external led that also connected to GPIO16 turn on and turn off in sync with my internal red led?

My brain can not process this logic.  When i set my external led off, my internal led is on... and they both can not do what i say. pinMode(16, OUTPUT) and digitalWrite(16, LOW) also not working, i mean, may be i can turn the external led on just like the internal one by connecting the external led anode to 3.3v rail and cathode to pin 16 with that piece of code, but... its just not working.

Please help me.

NarinLab, 2017-09-15:
I'm sorry, problem solved. My imperfect sensor in my face didnot see the NodeMCU pin is not really click to breadboard. This silly things make me crazy for six hours.

The code is right, just need to push the Node to breadboard. hahh....

mharmon12, 2017-09-15:
Best fix ever!
You have learned more than you know from this exercise.
Thank you for updating.

Raincode, 2019-04-20:
Thank you! I spent waaaay to long being confused as hell before googling it...

amitabheer, 2019-08-22:
Dear All,
Greetings and thanks to everyone's contribution here.
Kindly acknowledge that I too have facing this kind of problem with NODE MCU Lolin V3.
I have made a project on 8_Ch relay control via Alexa (via Sirin and Alexa). Everything is working well but all outputs are inverted. After uploading program to MCU the default output in ON (HIGH) for all 8 Channels (Pins 5,4,0,2,14,12, 13 and 15).
Even after lot of thinking I could not understand why its happening to this and how can the same be resolved?
I think this output can be inverted by changing (inverting) the Outputs 0 to1 and vice-versa but this doesn't seem good to me as the MCU (and my circuit as well) should behave as it was designed for.
So, all of you are requested to suggest me to attain the solution.
Thanks and regards!
-Dr Amit

vikrantsingh47, 2020-05-19:
found this article which explains it in detail: https://www.dataq.com/blog/data-acquisition/whats-all-this-sink-and-source-current-stuff/

lokaero, 2021-03-22:
> Hello!
>
> Let me get this straight. I bought the following board: [...]

lokaero, 2021-03-22:
After searching about this for a while.. i found the reason for this kind of behaviour in ESP8266 board..
The problem is When we give digital-write HIGH the output is LOW.. and LOW is given out as HIGH.

Solution for this problem is....
The pins highlighted in green are OK to use as output  and input(RECOMMENDED)... The ones highlighted in yellow are OK to use, but you need to pay attention because they may have unexpected behaviour mainly at boot this is also the reason why ESP8266 behaves inverted to digital in and out and reboots automatically after sometimes. The pins highlighted in red are not recommended to use as inputs or outputs.

So please use the correct GPIO pins for input and output..try to use pins that are marked green only....thank you..

[image]

mrnams, 2021-09-17:
Once we give low input to pin 16 of NodeMCU ESP8266, it remains low, see my code below which is not able to switch off led because of this issue, can't we use pin 16 as general purpose input?
See my complete code
https://github.com/mrnams/IOT/blob/main/ESP8266/Examples/Basic/One-Led-Two-Push-Buttons

chess-levin, 2022-01-14:
Hi guys, have a look at this great post https://www.instructables.com/ESP8266-Using-GPIO0-GPIO2-as-inputs/
I think it is great explanation of this special topic at the GPIO pins 0,2,15 of ESP8266 and how to use them and how you can still use them.