nodemcu/nodemcu-devkit-v1.0 #17: ESP8266 destroyed

CipTGCN, 2016-05-15:
Hi,

I have managed to burn 4 ESP8266 on a dev. board (https://github.com/nodemcu/nodemcu-devkit-v1.0), and the reason is still unknown for me.

The history is like this:
After I have burned the:
– 1st one I removed GPIO10 from my code an also let if floating,
– 2nd I have reduced the current for driving MOC3041 from 15 to 7.6mA,
– 3rd I removed the ADC In setting (analogRead) from my code, even if I don
t use it for anything, yet.

Now I have the following circuit:
220V AC -> Power transformer from 220 to 6V (AC) -> Rectifier bridge (Vout ~8.6V DC) -> 2 capacitors (470uF + 100nF) -> LM7805 -> Power the ESPshield to Pin 15 (+) and Pin 7 (-) -> use Pin8/GPIO14 as an output to power an opto-triac MOC3041 (current was limited to 7.6mA using a 270ohm resistor) ->a triac to drive a load of 21W (7 x AC LED lamp) -> Pin1/GPIO16 used as an input (if pin state is changed than also the output is changed).

After running the application ~15 minutes (Playing on/off via web requests) I have played also a little bit with the input pin and everything worked as expected. Then I have decided to let lamps ON (input pin was pulled to GND - Red led from shield lit), but after ~5 -10 minute the lamps turned off (Red led from shield remained on) and now the 4th ESP is dead.

I have checked the value of current limiting resistor for the Red Led (470 ohm, value was obtained after I have removed the resistor from the shield) and from my calculations the current ~3.2mA is small enough not to damage the ESP, and now I have no idea what I
m doing wrong.
If you think that is relevant I can provide also the code (written using Arduino IDE)

Have anyone an idea that can help me with this issue?

iandrosov, 2016-05-23:
What in your description indicates that you "Destroy ESP8266"? One thing I see in your schematic, you are using 5V to power NodeMCU, form what I read in docs recommended power is 3V although there is a pin-out you seem to be using that can run this unit at 5Volts I would try to reconfigure to use 3V or USB power to test. Unless you have requirement to run this at 5V.

CipTGCN, 2016-05-23:
The value of the power supply is in the safe range:

[image]

supersjimmie, 2016-09-07:
No, as far as I can see he connects +5V to the 3.3V pin, not to the 5V pin.
The 5V pin (VDD 5V) goes through the onboard 1117-3.3V regulator but the 3.3V pin he is using goes straight into the ESP chip. The chip will die because 5V on it's 3.3V pin is too much.

CipTGCN, 2016-09-07:
From schematic PIN 15 is 5V VDD
[image]

https://github.com/nodemcu/nodemcu-devkit-v1.0/issues/17#issuecomment-245269701

fiedlfa, 2016-09-07:
From your Schematics it looks a bit that you don't Connect Ground of the NodeMCU. Is this an Error in your Schematics?

CipTGCN, 2016-09-07:
My block diagram is not 100% precise..the GND pin used is pin 7 from the "right" side.
I have to say that all the time the entire pcb was near (5-10 cm) a bunch of wires powered with 230V  AC  (@ 60Hz)