nodemcu/nodemcu-devkit-v1.0 #18: North American Wholesaler

Jonathan411, 2016-09-01:
This is a non-technical issue, but may be an issue that others are encountering.

Our company, located in Canada, would like to make volume purchases of "official" (i.e. Amica) NodeMCU V2 modules.  However, we have been unable to locate a Canadian, or even US-based, wholesaler.

Can anyone provide contact information for a North American wholesaler of "real" Amica modules? (we don't want to purchase "knock-offs" for use in commercial products).

Thank you.

marcelstoer, 2016-10-12:
@nodemcu @vowstar @GerwinJanssen can you help out Jonathan?

Jonathan411, 2016-10-12:
Thanks Marcel, but I have located a Chinese supplier that can supply NodeMCU dev kits in volume, for less than $3 US per piece.  If you, or anyone you come across, is looking for a volume supplier of these units let me know and I will forward the Chinese contact information (contact was made via Alibaba.com).

 

FYI:  we are using the NodeMCU dev kit, instead of just the ESP8266 module, because we can’t build/assemble the power circuitry, CP2102, micro USB connector and ESP8266 module for anything close to $3 in North America. It’s an incredible platform for $3 - we are just soldering the entire dev kit onto our custom pcb (along with other components) to create the final product.

 

Thanks again,

 

Jonathan

themendes, 2016-10-13:
@Jonathan411 I am certainly interested. Could you please put me in touch with the relevant person? US$3 is remarkably cheap.

Jonathan411, 2016-10-13:
Hi,

 

To get the $3 unit price we are placing orders for thousands of units at a time, on a monthly basis, and we are also purchasing (by the thousands) cables and power supplies.  If you have a requirement for thousands of units at a time send me your contact information and I will forward it to our supplier.  If your requirement is for a lower volume, send me your estimated requirements and I will ask our supplier if they will handle smaller orders.

 

Jonathan

vowstar, 2016-10-13:
@Jonathan411 If you need NodeMCU devkit or modules, you can contact the factory of AiThinker(http://www.ai-thinker.com) in China, They produce all kinds of ESP modules and boards. They also have agents around the world.

Jonathan411, 2016-10-13:
Thanks for the link!

 

Jonathan

Jonathan411, 2016-10-13:
If you are looking for smaller volumes AI Thinker uses the following online store:

 

https://world.taobao.com/search/search.htm?_ksTS=1476361384777_41 <https://world.taobao.com/search/search.htm?_ksTS=1476361384777_41&spm=a21bp.7806943.20151106.1&_input_charset=utf-8&navigator=all&json=on&q=nodemcu&callback=__jsonp_cb&cna=iRadD11YxXwCAUxFEAlO84v9&abtest=_AB-LR517-LR854-LR895-PR517-PR854-PR895> &spm=a21bp.7806943.20151106.1&_input_charset=utf-8&navigator=all&json=on&q=nodemcu&callback=__jsonp_cb&cna=iRadD11YxXwCAUxFEAlO84v9&abtest=_AB-LR517-LR854-LR895-PR517-PR854-PR895

 

Prices for nodemcu modules seem very reasonable.

 

Jonathan

tablatronix, 2017-03-23:
I wish you could buy them without headers soldered, like other dev kits do.