nodemcu/nodemcu-devkit-v1.0 #19: Eagle library for Nodemcu Lolin V3

irwanayeah, 2016-10-16:
Hello gentleman,
I would like to use Nodemcu Lolin V3 as a part in my PCB layout. But I can't find any library for Eagle. Is there any existing one? Or does anybody plan to create one?
Thanks.

sturze, 2017-08-11:
me too! still nothing... nowhere...

vowstar, 2017-08-11:
@irwanayeah @sturze Hello, Lolin V3 is not official board, the official latest board is nodemcu-devkit-v1.0
And, we don't authorization Lolin V3 use the name Nodemcu.

closed 2017-08-11 by vowstar

Eltronas, 2017-11-08:
https://mega.nz/#!1t8jzSqb!doaTLDcj8MVNFQ1ReJRFOIpOO0-HCy4FIjuXTIL_IVw

I create one, I hope I've helped :)

cmedinam, 2018-01-22:
Eltronas I thing your package is wrong, Its weight is wrong or there are diferent sizes of the "same" board.

shasan247, 2018-01-29:
Eltronas, do you have any library for V2???

patavium73, 2018-03-04:
guardate questo che funzina https://www.arduinotronics.com/nodemcu-lolin-v-3-eagle-library/

HADIUU07, 2024-01-07:
> guardate questo che funziona https://www.arduinotronics.com/nodemcu-lolin-v-3-eagle-library/

can you sent link again ? the link expired