nodemcu/nodemcu-devkit-v1.0 #29: Connect relay board to NodeMCU

alexartd, 2017-11-15:
Hello, I've a relay board with 3 pins, VCC, GND and IN. I've connected VCC to 3.3 and GND to GND of MCU. In is connected to a GPIO pin. With pinMode(pin, OUTPUT); and digitalWrite(RELAY1,LOW); and digitalWrite(RELAY1,LOW); relay doesn't work. I've tested that if I connect relay IN pin to VCC it doesn't work but if I connect it to GND it work. This relay hasn't jumper to set if it should work in HIGHT or LOW. I suppose that MCU GPIO should provide a GND signal to turn on relay but I don't know how.
Can you help me ?

mB-PiBox, 2017-11-15:
Probably not supplying the correct pinout. Also you would need to define "RELAY1" to the pin you're using.

Relay works with a ON/OFF. Supplying power to pin X connected to "IN" turns relay off if you have pin set to output

alexartd, 2017-11-15:
If I connect led anode (+) to the same GPIO and led catode (-) to GND it works so GPIO is controlled correctly, but if the same GPIO is connected to IN pin of relay, it doesn't work.

mB-PiBox, 2017-11-15:
http://bfy.tw/F33K

alexartd, 2017-11-15:
Many thanks but it is what I've done and I'm sure that relay is connected well, as explained on internet. You haven't understand my question. I think that the problem is that this relay board work with low signal, GND to be clear. But a GPIO is not able to send a GND signal bit a positive signal from 0 to 3.3V. I think that with a transistor I can resolve but I'm searching other solutions.

mB-PiBox, 2017-11-15:
No, relays work the same. You have POWER or NO POWER to the input/trigger. Which Opens or Closes the gate/circuit depending on the NC/NO state of the relay.

alexartd, 2017-11-15:
Ok but I've wrote that if I connect a led in the place of relay, it work so GPIO work.

alexartd, 2017-11-15:
I think this is the solution:
https://iot-playground.com/images/articles/053/ESP8266_4_relay_switch.png

mB-PiBox, 2017-11-15:
Again, relay works the same. I have a 8 RELAY system that controls the lights in my work shop off a NodeMCU, and each relay input/trigger is connected to a GPIO. No need for any resistors (or in the case of the link you provided NO transistor) to make the relay work. 

I even  have a 2 channel relay sitting right here, that I just tested on a spare NodeMCU, and again no resistor or transistor needed.

alexartd, 2017-11-15:
Probably my relay is different from your. I don't know.

mB-PiBox, 2017-11-15:
THEY ALL WORK THE SAME! FFS

alexartd, 2017-11-18:
My relay turn on with pin IN connected to GND or also to 3.3V. GPIO pin from MCU turn from 0V to 3.3V but I don't understand why it cannot turn on relay. I've solved with circuit with transistor.
Bye