nodemcu/nodemcu-devkit-v1.0 #30: Gerber file latest version

manibenwa, 2017-12-19:
Hai ,

We are developing Nodemcu board and this github has 4 gerber files related to the Nodemcu board
which one should i select or which is the  latest version gerber file???

vowstar, 2018-01-02:
It is better to use the PCB source file, not gerber file. I think the latest gerber file is in master branch of this reop, but the most latest gerber file you'd better generate from PCB source file.

manibenwa, 2018-01-02:
Thanks for replay,
Where are the PCB source files ?? I haven't found the PCB source files in https://github.com/nodemcu/nodemcu-devkit-v1.0 link!!
Please help me....

vowstar, 2018-01-03:
https://github.com/nodemcu/nodemcu-devkit-v1.0/blob/master/NODEMCU_DEVKIT_V1.0.PcbDoc

This file is the PCB source file, please use Altium designer to open it.
All this project is open source and in this repo.

manibenwa, 2018-01-03:
Thanks VOWSTAR

manibenwa, 2018-01-10:
Hai,
           I found the BOM of this board in this link  https://github.com/nodemcu/nodemcu-devkit-v1.0/blob/master/NODEMCU_DEVKIT_V1.0.PDF. I need voltage rating of capacitor and wattage rating of resistors.could u tell me is there any possible way to get the ratings of this capacitors and resistors????

vowstar, 2018-01-10:
The parameters we actually use is

|Designator|parameter|
|-|-|
|C1,C4,C6 |100nF (104) ±10% 16V|
|C2 |100uF(107) ±10% 6.3V Case B 3528|
|C3,C5,C8 |10uF(106) ±20% 10V |
|C7|10uF(106) ±10% 25V|
|D1|SOD-123 40V,1A,VF=0.45V@1A|
|LED1|1.6x0.8x0.6 Iv= 35~65mcd @IF=5mA ESD:1000V |
|R1,R2,R3,R4,R5,R7,R8|12KΩ (1202) ±1% 0.0625W [50V TYP] [100V MAX] T.C.R ±100|
|R6,R9,R11,R12|470Ω (4700) ±1% 0.0625W [50V TYP] [100V MAX] T.C.R ±100|
|R10(Do not install)|0Ω (0R00) ±1% 0.0625W [50V TYP] [100V MAX] T.C.R ±100|
|R13|220kΩ (2203) ±1% 0.0625W [50V TYP] [100V MAX] T.C.R ±100|
|R14|100kΩ (1003) ±1% 0.0625W [50V TYP] [100V MAX] T.C.R ±100|

manibenwa, 2018-01-10:
Thanks vowstar.
if u have part numbers it is really help for me.

manibenwa, 2018-01-10:
Hai,
 does nodemcu have next version??if their send me some information ..

vowstar, 2018-01-11:
@manibenwa I'm taking part in the design of an AI chip, next open hardware project may be something about AI chip.

manibenwa, 2018-01-24:
Hai vowstar,
        Iam using NodeMCU DEVKIT V1.0.
        I would like to power the nodemcu board without usb. I want power up the  board  with battery   
       (5v or 12v).is vin pin is used to power the board with external battery ??