nodemcu/nodemcu-devkit-v1.0 #31: proteus design for circuit design

Amanrazdan, 2017-12-23:
i was trying make circuit design with nodemcu with proteus software but couldn't find the include libraries to work with it in the proteus software. Can any anyone help.

alanrivetta, 2018-03-04:
Is there anything new about this?

I have a NODEMCU with Arduino IDE and I would like to simulate it in Proteus.
Is there any way to get it?

PriyankBasrani, 2018-05-16:
Can you send me library of NODEMCU alanrivetta??

maresfrios, 2018-05-21:
Can you send me too library of NodeMCU ?
maresfrios@gmail.com  @alanrivetta

DavidHolo, 2018-10-09:
me too please ?  @alanrivetta or another else ?  @PriyankBasrani @maresfrios 
fikrifarhanrrasyid16@gmail.com

omarmhamza, 2019-07-20:
Here you go @DavidHolo @maresfrios @PriyankBasrani @alanrivetta @Amanrazdan 

http://www.mediafire.com/file/5zcbljfw9zbzc25/NodeMCU_for_Proteus_%2528Arduino_Magix%2529.rar/file

aasitvora99, 2021-08-01:
I guess we have to interface with an arduino and then run the code off arduino. Only then it'll work.
But if someone has any answer please do help out.