nodemcu/nodemcu-devkit-v1.0 #44: Pinout Documentation

mrleongalaxyum, 2020-11-15:
Hello,

I'm interested in software/template you used to generate the following pinout image:

[image]

I'm currently working on a LoRaWAN node device which will have quite a bit of headers so I'm interested in creating a similar pinout graphic.

Best regards,
Leon

vowstar, 2020-11-15:
[NODEMCU-DEVKIT-V1.0-INSTRUCTION-EN.zip](https://github.com/nodemcu/nodemcu-devkit-v1.0/files/5541772/NODEMCU-DEVKIT-V1.0-INSTRUCTION-EN.zip)

This was a very old story. In the past years I use M$ word. Now I'm using draw.io

mrleongalaxyum, 2020-11-15:
Thanks a lot!

closed 2020-11-15 by mrleongalaxyum